bolderflight/eigen #1: Conflicting identifiers with Arduino core

rblilja, 2021-12-20:
Eigen contains several symbols having conflicting identifiers with the Arduino-framework (core).

For an instance Eigen uses the identifier B1 giving the following errors: 

`.pio/libdeps/featheresp32/Bolder Flight Systems Eigen/src/Eigen/src/SVD/BDCSVD.h:378:12: error: request for member 'row' in '1', which is of non-class type 'int'
         B1.row(k1) = B.row(j);
            ^
.pio/libdeps/featheresp32/Bolder Flight Systems Eigen/src/Eigen/src/SVD/BDCSVD.h:389:55: error: request for member 'topRows' in '1', which is of non-class type 'int'
     A.topRows(n1).noalias()    = A1.leftCols(k1) * B1.topRows(k1);`

The above error is due to a #define of the identifier B1 in .platformio/packages/framework-arduinoespressif32/cores/esp32/binary.h (i.e. the Arduino core):

`#define B1 1`

Possibly there will be conflicts for A0, A1, B0 and 'round' as well: https://github.com/moritzmaier/Eigen/blob/master/Eigen.h

Unclear what the best solution would be.

flybrianfly, 2021-12-20:
That's bizarre, I renamed B1 throughout eigen to B1_ to avoid that conflict.  Can you ensure that you're using at least v3.0.1? I dislike the [undef approach](https://github.com/moritzmaier/Eigen/blob/master/Eigen.h) since there isn't any insight into how it's affecting the Eigen library, I'd rather go through the Eigen code and rename conflicts.

rblilja, 2021-12-21:
Agreed that undef is not the way to go.

It seems that the version you get when installing through Platform IO GUI is 3.0.0, also this webpage indicates version 3.0.0: https://platformio.org/lib/show/13236/Bolder%20Flight%20Systems%20Eigen

I'm not very knowledgable in how Platform IO works in regards of this.

Keep up the good work!

rblilja, 2021-12-21:
Managed to point Platform IO directly to this repository (v3.0.2) instead of the mirrored (v3.0.0) at https://platformio.org/lib/show/13236/Bolder%20Flight%20Systems%20Eigen.

Compiles. Closing issue.

closed 2021-12-21 by rblilja