bolderflight/eigen #2: Will not compile on Arduino IDE with an STM32F103 (Bluepill)

fjdebruijn81, 2022-01-08:
This fails to compile the MPU9250 example using this library with either SPI or I2C examples.
It seems to be unhappy about the round macro .

Here is the log from the Compiler
Arduino: 1.8.19 (Linux), Board: "Generic STM32F103C series, STM32F103C8 (20k RAM. 64k Flash), Serial, 72Mhz (Normal), Smallest (default)"

In file included from /home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish.h:63:0,
                 from /home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/Arduino.h:30,
                 from sketch/spi.ino.cpp:1:
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/MathFunctions.h:1389:51: error: expected unqualified-id before 'const'
 inline EIGEN_MATHFUNC_RETVAL(round, Scalar) round(const Scalar& x)
                                                   ^
/home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish_math.h:123:35: note: in definition of macro 'round'
 #define round(x)                ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                                   ^
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/MathFunctions.h:1389:51: error: expected ')' before 'const'
 inline EIGEN_MATHFUNC_RETVAL(round, Scalar) round(const Scalar& x)
                                                   ^
/home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish_math.h:123:35: note: in definition of macro 'round'
 #define round(x)                ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                                   ^
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/MathFunctions.h:1389:51: error: expected ')' before 'const'
 inline EIGEN_MATHFUNC_RETVAL(round, Scalar) round(const Scalar& x)
                                                   ^
/home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish_math.h:123:35: note: in definition of macro 'round'
 #define round(x)                ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                                   ^
In file included from /home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/Core:172:0,
                 from /home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/eigen.h:36,
                 from /home/user/Arduino/libraries/Bolder_Flight_Systems_MPU9250/src/mpu9250.h:38,
                 from /home/user/Arduino/libraries/Bolder_Flight_Systems_MPU9250/examples/arduino/spi/spi.ino:26:
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/GenericPacketMath.h: In function 'Packet Eigen::internal::pround(const Packet&)':
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/GenericPacketMath.h:826:48: error: 'Eigen::numext::round' has not been declared
 Packet pround(const Packet& a) { using numext::round; return round(a); }
                                                ^
In file included from /home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish.h:63:0,
                 from /home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/Arduino.h:30,
                 from sketch/spi.ino.cpp:1:
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/arch/Default/Half.h: At global scope:
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/arch/Default/Half.h:732:50: error: expected unqualified-id before 'const'
 EIGEN_STRONG_INLINE EIGEN_DEVICE_FUNC half round(const half& a) {
                                                  ^
/home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish_math.h:123:35: note: in definition of macro 'round'
 #define round(x)                ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                                   ^
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/arch/Default/Half.h:732:50: error: expected ')' before 'const'
 EIGEN_STRONG_INLINE EIGEN_DEVICE_FUNC half round(const half& a) {
                                                  ^
/home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish_math.h:123:35: note: in definition of macro 'round'
 #define round(x)                ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                                   ^
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/arch/Default/Half.h:732:50: error: expected ')' before 'const'
 EIGEN_STRONG_INLINE EIGEN_DEVICE_FUNC half round(const half& a) {
                                                  ^
/home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish_math.h:123:35: note: in definition of macro 'round'
 #define round(x)                ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                                   ^
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/arch/Default/BFloat16.h:564:54: error: expected unqualified-id before 'const'
 EIGEN_STRONG_INLINE EIGEN_DEVICE_FUNC bfloat16 round(const bfloat16& a) {
                                                      ^
/home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish_math.h:123:35: note: in definition of macro 'round'
 #define round(x)                ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                                   ^
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/arch/Default/BFloat16.h:564:54: error: expected ')' before 'const'
 EIGEN_STRONG_INLINE EIGEN_DEVICE_FUNC bfloat16 round(const bfloat16& a) {
                                                      ^
/home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish_math.h:123:35: note: in definition of macro 'round'
 #define round(x)                ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                                   ^
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/arch/Default/BFloat16.h:564:54: error: expected ')' before 'const'
 EIGEN_STRONG_INLINE EIGEN_DEVICE_FUNC bfloat16 round(const bfloat16& a) {
                                                      ^
/home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish_math.h:123:35: note: in definition of macro 'round'
 #define round(x)                ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                                   ^
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/functors/UnaryFunctors.h: In member function 'const Scalar Eigen::internal::scalar_round_op<Scalar>::operator()(const Scalar&) const':
/home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish_math.h:123:33: error: expected unqualified-id before '(' token
 #define round(x)                ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                                 ^
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/functors/UnaryFunctors.h:798:106: note: in expansion of macro 'round'
   EIGEN_DEVICE_FUNC EIGEN_STRONG_INLINE const Scalar operator() (const Scalar& a) const { return numext::round(a); }
                                                                                                          ^
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/../plugins/ArrayCwiseUnaryOps.h: At global scope:
/home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish_math.h:123:36: error: expected unqualified-id before ')' token
 #define round(x)                ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                                    ^
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/../plugins/ArrayCwiseUnaryOps.h:467:1: note: in expansion of macro 'round'
 round() const
 ^
/home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish_math.h:123:37: error: expected ')' before '>=' token
 #define round(x)                ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                                     ^
/home/user/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/../plugins/ArrayCwiseUnaryOps.h:467:1: note: in expansion of macro 'round'
 round() const
 ^
Multiple libraries were found for "Wire.h"
 Used: /home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/libraries/Wire
 Not used: /home/user/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/libraries/WireSlave
exit status 1
Error compiling for board Generic STM32F103C series.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

flybrianfly, 2022-01-08:
Thanks, I was able to reproduce this issue. Indeed the first error is with the _round_ macro in the STM32F103 cores. I quickly bypassed this issue by adding `#undef round` to the _eigen.h_ header file. Unfortunately, the STM32F103 core is using a very old toolchain (C++11), so it then has issues with the variable templates used in the _units_ library. There are workarounds for that (i.e. defining functions instead of constants), but they have performance implications, especially in embedded. This would be resolved by moving at least to C++14.

I'm not sure the best path to fix _eigen_ and _units_ to work with the STM32F103 or whether those fixes should be made. In the meantime, I suggest forking the _mpu9250_ library and removing the dependencies on _eigen_ and _units_. Sorry, I wish it was a more straightforward fix.

Cougar83, 2022-04-15:
Hello sir
I download this library, I installed it and I just ran the example to check if it was working. 
then I did the compilation and the IDE Arduino (2.0 RC5) returned me this message:

In file included from /Users/Documents/Arduino/libraries/eigen-main/src/Eigen/Core:19:0,
                 from /Users:Documents/Arduino/libraries/eigen-main/src/eigen.h:36,
                 from /Users/Documents/Arduino/libraries/eigen-main/examples/arduino/eigen_example/eigen_example.ino:5:
/Users/Documents/Arduino/libraries/eigen-main/src/Eigen/src/Core/util/Macros.h:679:10: fatal error: cmath: No such file or directory
 #include <cmath>
   

in fact I want to use it with this library : https://github.com/FlyTheThings/uNavINS#axis-system. The author is using your library (even give this address)  and after the compilation, the system is returning me also an error : 

In file included from /Documents/Arduino/Sketches/GPS/TinyGPSplus-for-Neo-M8N-master/examples/FullExample/FullExample.ino:1:0:
/Users//Documents/Arduino/libraries/uNavINS-master/uNavINS.h:48:12: fatal error: Eigen.h: No such file or directory
   #include "Eigen.h"

I am a new for sure, but I already use several time other libraries and now, with this problem, I m little bit lost :(

Thank you in advance for your time.
:)

flybrianfly, 2022-06-30:
@Heikkif, you can try v1.0.2 of the [MPU9250 Library](https://github.com/bolderflight/mpu9250/releases/tag/v1.0.2), which does not have dependencies on the Eigen or Units library.

FrankXu626522, 2023-12-18:
> @Cougar83, which processor are you using? Based on the error message, it seems like it's not an ARM processor.

i met the same error, im working on porting mini cheetah's code on arduino uno, lots of cpp .h or lib can't be found,

flybrianfly, 2023-12-18:
@FrankXu626522, are you porting to an Arduino Uno? I wouldn't expect this to work on that platform. It's going to need a 32 bit ARM processor to work.