bolderflight/eigen #4: Compile issue with different ARM Cortex-M0+ boards

gerowtj, 2022-08-21:
The example code compiles correctly on my computer using Arduino IDE 1.8.13 and the Teensy LC board.  The Teensy LC uses an ARM Cortex-M0+ processor.  The board I am trying to use for my project is also an ARM Cortex-M0+ made by Seeeduino, the [XIAO](https://wiki.seeedstudio.com/Seeeduino-XIAO/).

The Teensy LC compiles without any errors using your example program.  
Changing the board to the XIAO gives several pages of errors.  
Below are just the first few that have files for both the seeeduino board and eigen library listed:
```

In file included from sketch\sketch.ino.cpp:1:0:
c:\users\arduino\appdata\local\arduino15\packages\seeeduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\complex:69:30: error: expected unqualified-id before 'const'
   template<typename _Tp> _Tp abs(const complex<_Tp>&);
                              ^
c:\users\arduino\appdata\local\arduino15\packages\seeeduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\complex:69:30: error: expected ')' before 'const'
c:\users\arduino\appdata\local\arduino15\packages\seeeduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\complex:69:30: error: expected ')' before 'const'
c:\users\arduino\appdata\local\arduino15\packages\seeeduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\complex:601:5: error: expected unqualified-id before 'const'
     abs(const complex<_Tp>& __z) { return __complex_abs(__z); }
     ^
c:\users\arduino\appdata\local\arduino15\packages\seeeduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\complex:601:5: error: expected ')' before 'const'
c:\users\arduino\appdata\local\arduino15\packages\seeeduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\complex:601:5: error: expected ')' before 'const'
c:\users\arduino\appdata\local\arduino15\packages\seeeduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\complex: In static member function 'static _Tp std::_Norm_helper<true>::_S_do_it(const std::complex<_Tp>&)':
C:\Users\Arduino\AppData\Local\Arduino15\packages\Seeeduino\hardware\samd\1.8.3\cores\arduino/Arduino.h:137:16: error: expected unqualified-id before '(' token
 #define abs(x) ((x)>0?(x):-(x))
                ^
In file included from C:\Users\Arduino\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/Core:50:0,
                 from C:\Users\Arduino\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/eigen.h:36,
                 from C:\Users\Arduino\Documents\Arduino\EigenProj\sketch\sketch.ino:1:
c:\users\arduino\appdata\local\arduino15\packages\seeeduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\complex: In function 'std::complex<_Tp> std::__complex_log(const std::complex<_Tp>&)':
c:\users\arduino\appdata\local\arduino15\packages\seeeduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\complex:773:26: error: expected primary-expression before '(' token
     { return complex<_Tp>(log(std::abs(__z)), std::arg(__z)); }
                          ^
In file included from sketch\sketch.ino.cpp:1:0:
C:\Users\Arduino\AppData\Local\Arduino15\packages\Seeeduino\hardware\samd\1.8.3\cores\arduino/Arduino.h:137:16: error: expected unqualified-id before '(' token
 #define abs(x) ((x)>0?(x):-(x))
                ^
c:\users\arduino\appdata\local\arduino15\packages\seeeduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\complex: In function 'std::complex<_Tp> std::__complex_sqrt(const std::complex<_Tp>&)':
C:\Users\Arduino\AppData\Local\Arduino15\packages\Seeeduino\hardware\samd\1.8.3\cores\arduino/Arduino.h:137:16: error: expected unqualified-id before '(' token
 #define abs(x) ((x)>0?(x):-(x))
                ^
c:\users\arduino\appdata\local\arduino15\packages\seeeduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\complex: In function '_Tp std::fabs(const std::complex<_Tp>&)':
C:\Users\Arduino\AppData\Local\Arduino15\packages\Seeeduino\hardware\samd\1.8.3\cores\arduino/Arduino.h:137:16: error: expected unqualified-id before '(' token
 #define abs(x) ((x)>0?(x):-(x))
                ^
C:\Users\Arduino\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/MathFunctions.h: At global scope:
C:\Users\Arduino\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/MathFunctions.h:1389:51: error: expected unqualified-id before 'const'
 inline EIGEN_MATHFUNC_RETVAL(round, Scalar) round(const Scalar& x)
                                                   ^
C:\Users\Arduino\AppData\Local\Arduino15\packages\Seeeduino\hardware\samd\1.8.3\cores\arduino/Arduino.h:139:24: note: in definition of macro 'round'
 #define round(x)     ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                        ^
C:\Users\Arduino\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/MathFunctions.h:1389:51: error: expected ')' before 'const'
 inline EIGEN_MATHFUNC_RETVAL(round, Scalar) round(const Scalar& x)
                                                   ^
C:\Users\Arduino\AppData\Local\Arduino15\packages\Seeeduino\hardware\samd\1.8.3\cores\arduino/Arduino.h:139:24: note: in definition of macro 'round'
 #define round(x)     ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                        ^
C:\Users\Arduino\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/MathFunctions.h:1389:51: error: expected ')' before 'const'
 inline EIGEN_MATHFUNC_RETVAL(round, Scalar) round(const Scalar& x)

```

I am assuming these errors are related to the C libraries that the Seeeduino board uses in the arduino compiler, from the ..\\Seeeduino\hardware\samd\1.8.3\cores\ folder.  
Some of the errors show headers from the eigen library as well that I assume are because of the errors with the Seeeduino code.

Do you know if there is any work around for these?  My skill set is pretty limited in dealing with compilers and any core files for the boards.

Thank you

gerowtj, 2022-08-21:
I've spent some time looking at the differences between the Arduino.H file for the Teensy board and the Xiao board.  They both have a section to redefine abs but for some reason the XIAO does it twice, otherwise the code is the same:

```
// undefine stdlib's abs if encountered
#ifdef abs
#undef abs
#endif // abs
// undefine stdlib's abs if encountered
#ifdef abs
#undef abs
#endif // abs
```
followed later on in the code with a new define:
`#define abs(x) ((x)>0?(x):-(x))`

I commented out the whole section on ifdef abs and define abs.
```
// undefine stdlib's abs if encountered
//#ifdef abs
//#undef abs
//#endif // abs
// undefine stdlib's abs if encountered
//#ifdef abs
//#undef abs
//#endif // abs

//#define abs(x) ((x)>0?(x):-(x))
```
The XIAO arduino.h also has a redefine for round(x) which the Teensy does not:
`#define round(x)     ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))`

I commented that section out as well.
`//#define round(x)     ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))`

Making these changes I was able to compile the code and get the following results:

```
nrow: 6
ncol: 6

0.112775, 0.048292, -0.313197, -0.067979, 0.238543, 0.096648, 
-0.002733, -0.177438, 1.128801, 0.750399, -1.283369, -0.084223, 
0.645584, 0.189010, -1.758835, -1.095649, 2.438091, 0.718835, 
-0.613755, -0.359209, 2.150181, 1.179505, -2.575186, -0.930255, 
0.504729, 0.085257, -0.751615, -0.277679, 1.106807, 0.128982, 
-0.107398, 0.104756, 0.108780, 0.094344, -0.231374, 0.222546, 
```

I used a [website](https://wokwi.com/) that simulates arduino boards to test the code with the ArduinoEigen library using a simulated ESP32 and got the same results as the XIAO board using the Arduino IDE on my laptop.

So I believe this fixed the issue for me.

Can you confirm if the results are accurate?

Thanks

mjs513, 2022-09-13:
@flybrianfly -

Note these are the same errors that we are now getting with the Teensy3.x's and T4.x's upon switching tool chains to 11.3.1.  I did post about it on the forum:
https://forum.pjrc.com/threads/71074-Teensyduino-1-58-Beta-2?p=312709&viewfull=1#post312709

Would require a change to wiring.h to fix.