bolderflight/eigen #7: Basic Example Compile Error

PowerBroker2, 2023-04-24:
Compiled the basic Eigen.h example using the Arduino IDE (v2.1.0) for a Teensy 4.1 and got the following errors. Is there a way to fix this?

```
from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/eigen.h:36,
                 from C:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\examples\arduino\eigen_example\eigen_example.ino:5:
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1499:27: error: 'prefetch' is not a member of 'Eigen::internal'; did you mean 'Eigen::prefetch'?
 1499 |                 internal::prefetch(blB + (4 * K + 16) * RhsProgress);             \
      |                           ^~~~~~~~
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1526:13: note: in expansion of macro 'EIGEN_GEBP_ONESTEP'
 1526 |             EIGEN_GEBP_ONESTEP(1);
      |             ^~~~~~~~~~~~~~~~~~
In file included from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/Core:172,
                 from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/eigen.h:36,
                 from C:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\examples\arduino\eigen_example\eigen_example.ino:5:
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/GenericPacketMath.h:719:57: note: 'Eigen::prefetch' declared here
  719 | template<typename Scalar> EIGEN_DEVICE_FUNC inline void prefetch(const Scalar* addr)
      |                                                         ^~~~~~~~
In file included from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/Core:330,
                 from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/eigen.h:36,
                 from C:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\examples\arduino\eigen_example\eigen_example.ino:5:
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1497:25: error: 'prefetch' is not a member of 'Eigen::internal'; did you mean 'Eigen::prefetch'?
 1497 |               internal::prefetch(blA + (3 * K + 16) * LhsProgress);               \
      |                         ^~~~~~~~
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1527:13: note: in expansion of macro 'EIGEN_GEBP_ONESTEP'
 1527 |             EIGEN_GEBP_ONESTEP(2);
      |             ^~~~~~~~~~~~~~~~~~
In file included from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/Core:172,
                 from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/eigen.h:36,
                 from C:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\examples\arduino\eigen_example\eigen_example.ino:5:
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/GenericPacketMath.h:719:57: note: 'Eigen::prefetch' declared here
  719 | template<typename Scalar> EIGEN_DEVICE_FUNC inline void prefetch(const Scalar* addr)
      |                                                         ^~~~~~~~
In file included from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/Core:330,
                 from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/eigen.h:36,
                 from C:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\examples\arduino\eigen_example\eigen_example.ino:5:
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1499:27: error: 'prefetch' is not a member of 'Eigen::internal'; did you mean 'Eigen::prefetch'?
 1499 |                 internal::prefetch(blB + (4 * K + 16) * RhsProgress);             \
      |                           ^~~~~~~~
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1527:13: note: in expansion of macro 'EIGEN_GEBP_ONESTEP'
 1527 |             EIGEN_GEBP_ONESTEP(2);
      |             ^~~~~~~~~~~~~~~~~~
In file included from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/Core:172,
                 from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/eigen.h:36,
                 from C:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\examples\arduino\eigen_example\eigen_example.ino:5:
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/GenericPacketMath.h:719:57: note: 'Eigen::prefetch' declared here
  719 | template<typename Scalar> EIGEN_DEVICE_FUNC inline void prefetch(const Scalar* addr)
      |                                                         ^~~~~~~~
In file included from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/Core:330,
                 from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/eigen.h:36,
                 from C:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\examples\arduino\eigen_example\eigen_example.ino:5:
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1497:25: error: 'prefetch' is not a member of 'Eigen::internal'; did you mean 'Eigen::prefetch'?
 1497 |               internal::prefetch(blA + (3 * K + 16) * LhsProgress);               \
      |                         ^~~~~~~~
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1528:13: note: in expansion of macro 'EIGEN_GEBP_ONESTEP'
 1528 |             EIGEN_GEBP_ONESTEP(3);
      |             ^~~~~~~~~~~~~~~~~~
In file included from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/Core:172,
                 from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/eigen.h:36,
                 from C:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\examples\arduino\eigen_example\eigen_example.ino:5:
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/GenericPacketMath.h:719:57: note: 'Eigen::prefetch' declared here
  719 | template<typename Scalar> EIGEN_DEVICE_FUNC inline void prefetch(const Scalar* addr)
      |                                                         ^~~~~~~~
In file included from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/Core:330,
                 from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/eigen.h:36,
                 from C:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\examples\arduino\eigen_example\eigen_example.ino:5:
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1499:27: error: 'prefetch' is not a member of 'Eigen::internal'; did you mean 'Eigen::prefetch'?
 1499 |                 internal::prefetch(blB + (4 * K + 16) * RhsProgress);             \
      |                           ^~~~~~~~
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1528:13: note: in expansion of macro 'EIGEN_GEBP_ONESTEP'
 1528 |             EIGEN_GEBP_ONESTEP(3);
[... 115 lines cut ...]
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1499:27: error: 'prefetch' is not a member of 'Eigen::internal'; did you mean 'Eigen::prefetch'?
 1499 |                 internal::prefetch(blB + (4 * K + 16) * RhsProgress);             \
      |                           ^~~~~~~~
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1532:13: note: in expansion of macro 'EIGEN_GEBP_ONESTEP'
 1532 |             EIGEN_GEBP_ONESTEP(7);
      |             ^~~~~~~~~~~~~~~~~~
In file included from c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/Core:172,
[... cut ...]  int _MaxRows = -1; int _MaxCols = -1]'
C:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\examples\arduino\eigen_example\eigen_example.ino:47:40:   required from here
c:\Users\ltber\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/src/Core/products/GeneralBlockPanelKernel.h:2464:21: error: 'ptranspose' was not declared in this scope
 2464 |           ptranspose(kernel);
      |           ~~~~~~~~~~^~~~~~~~

exit status 1

Compilation error: exit status 1
```

flybrianfly, 2023-04-25:
It works with Arduino 1.8.x if you're able to use that. Not sure what's going on with Arduino 2 yet.

PowerBroker2, 2023-04-25:
Yeah, I've been using 1.8 up until today and had no issues with Eigen. Too bad it doesn't work with 2.1 yet - seems like a much improved IDE. If/when you figure out what's going wrong with 2.1 and Eigen, can you let know via this issue? Thanks!

PowerBroker2, 2023-04-28:
FYI, I reverted to Arduino 1.8.X and am still getting the errors...

flybrianfly, 2023-04-28:
Which version of Teensyduino?

PowerBroker2, 2023-04-28:
Teensyduino version: 1.58
Arduino version: 1.8.19
OS: Windows 10

flybrianfly, 2023-04-28:
Thanks, I was using Teensyduino 1.57. Upgrading to Teensyduino 1.58 reproduces the errors that you're seeing.

flybrianfly, 2023-04-28:
Seems like Teensyduino 1.59 should fix this, but it's still in beta. Otherwise, Teensyduino 1.57 works also.

PowerBroker2, 2023-04-28:
Ok, I'll give those a shot when I get the chance - thanks!

flybrianfly, 2023-04-28:
I think 1.59-beta3 is planned to have the fixes in it. Otherwise 1.57 should work fine.

PowerBroker2, 2023-04-28:
Figured out how to roll back Teensyduino to 1.57 and it all works again, thanks!

closed 2023-04-28 by PowerBroker2